PX4/PX4-ECL #1031: ekf terrain process

garlinplus, 2021-09-22:
When quadrotor fly cross step surface, ranger measure a gap.When in ranger fusion mode,how to process this situation to make measured height no jump point?

bresch, 2021-09-22:
That's exactly the reason why using range finder as primary height source is only recommended when the terrain is flat (e.g.: warehouse). With a good accelerometer you might be able to reject the step and assign it as a terrain change, but it's easier if you can use baro or GNSS height fusion.

garlinplus, 2021-09-23:
> That's exactly the reason why using range finder as primary height source is only recommended when the terrain is flat (e.g.: warehouse). With a good accelerometer you might be able to reject the step and assign it as a terrain change, but it's easier if you can use baro or GNSS height fusion.
In ranger fusion mode,we can use fault detector to detect ranger gap,and reset ranger offset to keep height no jump.

bresch, 2021-09-23:
> In ranger fusion mode,we can use fault detector to detect ranger gap,and reset ranger offset to keep height no jump.

Do you mean to use the innovation test ratio to detect a step? Yes, that's possible if you have a good accelerometer that you can trust and that the range finder is reliable too; otherwise it will be difficult to detect those steps.

garlinplus, 2021-09-23:
> Do you mean to use the innovation test ratio to detect a step? Yes, that's possible if you have a good accelerometer that you can trust and that the range finder is reliable too; otherwise it will [...]

yes,when rng test ratio > 1，we can reset ranger offset to keep ranger mesure no jump.
Accelerometer integrate bettween two ranger measure time,the drift is not  too big.

garlinplus, 2021-09-24:
@bresch Can we use baro as main height fusion sensor,and use ranger fusion to estimator baro drift.