PX4/PX4-ECL #1032: EKFGSF ahrsAlignYaw does not affect filter state

rjgritter, 2021-11-12:
In EKFGSF_yaw.cpp, we have the following function:

```
void EKFGSF_yaw::ahrsAlignYaw()
{
	// Align yaw angle for each model
	for (uint8_t model_index = 0; model_index < N_MODELS_EKFGSF; model_index++) {
		Dcmf& R = _ahrs_ekf_gsf[model_index].R;
		const float yaw = wrap_pi(_ekf_gsf[model_index].X(2));
		R = updateYawInRotMat(yaw, R);

		_ahrs_ekf_gsf[model_index].aligned = true;
	}
}
```

Is it just me, or does all this math do nothing to alter the state of the filter? A rotation matrix is updated but never pushed back to the filter.

I presume `R` of each model is intended to be updated to the newly-calculated `R`?

bresch, 2021-11-12:
```
Dcmf& R = _ahrs_ekf_gsf[model_index].R;
```
Means that `R` is a reference (note the `&`) to `_ahrs_ekf_gsf[model_index].R`, a rotation matrix of `_ahrs_ekf_gsf`, which is a member variable. Then, `R = ` is actually updating `_ahrs_ekf_gsf[model_index].R`.

rjgritter, 2021-11-12:
Thanks, I read through that too quickly!

closed 2021-11-12 by rjgritter