PX4/PX4-ECL #1034: Ekfgsf_yaw in rtk gps

iceberg1369, 2022-03-09:
Does EKFGsf_yaw can converge when using a rtk gps  instead of spp mode?  Is a few centimeters horizontal movement enough to converge?

bresch, 2022-03-09:
> Does EKFGsf_yaw can converge when using a rtk gps instead of spp mode?

Yes, absolutely, the better the speed accuracy, the faster the convergence.

> Is a few centimeters horizontal movement enough to converge?

There is no "yes" or "no" clear answer to that. The convergence time depends on the yaw observability and the yaw observability depends on the accelerometer noise, GNSS velocity accuracy and the acceleration of the drone.
So if you have high speed accuracy and a large acceleration with high SNR, then the filter can converge really quickly.
Best is to do some experiments with your setup. You can then fine tune the parameters of the yaw estimator to get the performance you need.

bresch, 2022-03-09:
Note that EKF2 moved into [PX4-Autopilot](https://github.com/PX4/PX4-Autopilot/tree/master/src/modules/ekf2/EKF), this repository is deprecated. For further questions, please join the PX4 forum: https://discuss.px4.io/

closed 2022-03-09 by bresch