PX4/PX4-ECL #1035: External vision yaw being rejected for portions of the flight

marcelino-pensa, 2022-03-11:
Hi,

I am flying px4 1.12.3 with mavros 1.13.0 providing external position and heading from a T265 module.

In one of my flights I had noticed that px4's heading drifted apart from the external vision yaw (blue plot has external vision heading, red plot has px4 heading):

[image]

I went ahead to inspect the ulog, and I have noticed that the flag `reject_yaw` did turn to `true` midway through this flight:

[image]

Then I checked the heading innovation, and it does shoot up quickly, then it slowly moved back down, never reaching a point where it is low enough to allow external vision yaw to be used:

[image]

Now I'm wondering if there is any way to disable this innovation test. I want the estimator to always trust external pose. What I imagine is that I need to increase some standard deviation parameters. I looked into the EKF code, and the variable for `reject_yaw` only shows up in `gps_yaw_fusion` and `mag_fusion`, but that is confusing me, as I'm not using either of those.

My question is: can I disable such test? Should I just change the standard deviation for one of these parameters? Which one? I could only find `EKF2_HDG_GATE` (which is supposed to be used with a magnetometer) and `EKF2_MAG_GATE` (also for magnetometer).

Note: my current value for `EKF2_AID_MASK` is 280.

marcelino-pensa, 2022-03-11:
Oops, forgot to upload the ulog. Here it is:
[log_257_2022-3-11-13-35-38.zip](https://github.com/PX4/PX4-ECL/files/8235965/log_257_2022-3-11-13-35-38.zip)

Xiaoye-A, 2022-10-11:
您好，奎已经收到你的邮件，谢谢您的来信哦！奎会尽快给您回复。