PX4/PX4-ECL #1040: EKF not tracking external vision position sent through MAVROS

s-niket, 2023-04-21:
Hi, 

I am running the version v1.13.2 of PX4 on my quadcopter, communicating to an offboard computer via MAVROS. 

I send position and orientation estimates to PX4, generated from a lidar. The EKF2_AID_MASK is set to 24. 

To validate my setup, I take out the props from the drone, I arm the drone and pick it up and move it around. I notice that the local_position estimates of PX4 drift away from the vision estimates after a while. I have noticed in the ROS bags that the vision yaw estimate doesn't match the EKF estimate and hence it starts diverging but I could be wrong. I believe the transformations are correct. 

[image]

Attaching the flight log [here](https://review.px4.io/plot_app?log=15fbd617-ca79-4231-a2e6-62e8f582144e)

Is my test valid? Or is there a better way of testing this? 

Any help is appreciated towards mitigating this issue. Thanks

bresch, 2023-04-25:
The innovation is 0 during the whole log, meaning that the EKF isn't even trying to fuse the data.
[image]
The data is most likely never "ready": https://github.com/PX4/PX4-Autopilot/blob/504d951c75fda334ba6ad0691cef83e8a66e5af4/src/modules/ekf2/EKF/control.cpp#L165

Could you try in v1.14 or `main` if you have the same issue (ev fusion has been completely rewritten)?

s-niket, 2023-05-05:
I think I found the issue. I had to change the timestamp just before publishing the data to MAVROS. That seemed to have made a huge difference, now the PX4 estimates mostly follow the EV estimates.

closed 2023-05-06 by bresch